touchgadget/esp32-usb-host-demos #1: Midi host

marchingband, 2021-11-23:
This is really cool!  Are you open to contributions? I might be able to hook up a parser to get this closer to a full working implementation

touchgadget, 2021-11-23:
I suggest starting your own repo because I will not have time to review and merge code. I do not mind anyone using code from this repo for other projects. MIDI is not a high priority for me at this time.

I received some ESP32 S3 boards so will be moving off to ESP-IDF S3 world for a while.

marchingband, 2021-11-23:
Ok cool, I will. Let me know if/how you would like credit!
S3 is in my future too, I am hoping the USB API is similar.

touchgadget, 2021-11-24:
In your README add an acknowledgement and link back to this repo.

marchingband, 2021-11-24:
Ok will do!

csBlueChip, 2021-11-26:
@marchingband 
If I can get this working (hopefully over the weekend), I have a LOT of parser code ready to be tested [*currently* here](https://github.com/csBlueChip/tinyusb/tree/master/_MIDIH.tmp)  ...Happy to work with you, or feel free to steal any useful logic (CC lookup tables, erroneous PID/VID list, etc.) - all MIT: "Share and Enjoy"

marchingband, 2021-11-26:
@csBlueChip i have had a lot of luck with midiXparser, used it in 2 projects with great success, I was planning to just add that!

marchingband, 2021-11-26:
https://github.com/TheKikGen/midiXparser

csBlueChip, 2021-11-26:
That is a stunningly clean bit of code, but only seems to handle a few basic /input/ commands ...I'm aiming for something a bit involved: {SMF player, full GM/GM2 synth, virtual keyboard} ...I also want to track RPN and NPN values for features such as pitch bend, as well as maintain routing tables between multiple USB & GPIO/UART ports.
...Still looking forward to seeing your project - something clean and functional is often more impressive than "feature bloated"

marchingband, 2021-11-26:
That’s very ambitious, wonderful, I can’t wait to play with it!

copych, 2023-12-08:
@csBlueChip @marchingband please, have you managed to develop s2/s3 midi host ability? TinyUSB still says there's no host support ((( Am I getting it somehow wrong, maybe there's some alternative way or did they implement it in the core libs already? Please, help, I just want to connect a keyboard to ESP32s3 ))).

marchingband, 2023-12-08:
it looks like support was merged very recently, so the way I did it is probably not relevant anymore.
https://github.com/hathach/tinyusb/releases/tag/0.16.0

marchingband, 2023-12-08:
But for completeness, here is a repo that helped me, I have not published my code yet. https://github.com/intechstudio/knot

copych, 2023-12-09:
> it looks like support was merged very recently, so the way I did it is probably not relevant anymore. https://github.com/hathach/tinyusb/releases/tag/0.16.0

Thank you for the response! But I cannot find where they mention ESP32 s2/s3 host support.

This is where I wanted to add USB MIDI kbd support: https://github.com/copych/AcidBox . Can you tell in advance, if USB host support is CPU-expensive?

Can you also tell, please, will any of the solutions work with USB MIDI keyboard connected via an USB-hub? Or do they only support a single device connected directly?

Looking at WVR I can imagine that you have made a long way from an idea to the product. Some time ago I haven't found your WVR project and started thinking of creating something similar, but with no dedicated firmware. My measurements has shown the ability of playing 17 simultaneous stereo samples directly from an SD-card connected via 4-bit bus, but required precaching first hundreds of bites from each sample of a set. Do you use similar technique or does a soldered memory chip solve these lag issues in a simple and efficient manner?

Also I have an idea of making a hybrid drum synth combining FM, wave-table and subtractive synthesis along with some basic FX in a chain. To keep real-time calculations within esp32's capacity I thought on pre-rendering some massive pieces and play them as samples, yet having live control of biquad or similar filtering and other fx. How do you think, may this be interesting to the community, and does it fit your hardware if I get it right at all? ))

marchingband, 2023-12-09:
I don't know if they support usb hub, I have not tried. WVR does not pre-cache data, the eMMC is fast enough when reading contiguous data. DSP is too slow on ESP32 when using floats, but can be done in fixed-point, I have not done much yet myself, it is hard. I suggest you order or manufacture a WVR and experiment, the hardware is open source. You can ask questions in the WVR forum.

copych, 2023-12-09:
Regarding float processing, I'd suggest you to check my AcidBox repo, it only requires ESP32 or RSP32S3 with PSRAM and preferrably some I2S DAC. Two cores are able of running 2 oscillators with filters, drum sampler with djgfilter, plus fx-chain consisting of overdrive, distortion,  reverb, delay and compressor. All @44100 sampling rate.

01GOD, 2024-08-20:
> This is really cool! Are you open to contributions? I might be able to hook up a parser to get this closer to a full working implementation

Can post a functional edition of that for S3 (and other compatible USB MIDI Host ESP32) n a separate repo from wvr? Would truly appreciate a functional USB MIDI Host solution, and I know more in the community would also. 

Thanks in advance, if possible!

marchingband, 2024-08-20:
@01GOD WVR uses a Seeed XIAO D21 and this code https://github.com/marchingband/wvr_usb_backpack
I also sell this as a standalone module here: https://www.ultrapalace.com/product-page/midicrown

01GOD, 2024-08-21:
> @01GOD WVR uses a Seeed XIAO D21 and this code https://github.com/marchingband/wvr_usb_backpack I also sell this as a standalone module here: https://www.ultrapalace.com/product-page/midicrown

Ah. Had thought were doing direct USB MIDI to that ESP32 instead of using another module for that. 
By the way, would be nice if can post a "THIS DEFINITELY FUNCTIONS for USB MIDI from laptop DAW to ESP32-S3 (and other compatible ESP32 boards) synths in Arduino IDE" example .ino. I know that people have been asking. I had installed the fourtyseveneffects Arduino MIDI Library yesterday, but there didn't seem to be a USB MIDI example for ESP32-S3 (and other compatible ESP32) in that. Surprisingly, even Marcel License ESP32 synths still rely on a separate USB MIDI solution. 

If post a simple example of that (and maybe add a saw wave synth later), that would likely be a very popular repo and that could be useful marketing for advertising wvr.