touchgadget/esp32-usb-host-demos #3: After a simple modification in usbhhelp.hpp, your library can work with ESP32 S3

yanbwang, 2022-06-18:
Hi,

Thanks for your works. And I have got a ESP32 S3 these days. With your original library, there would be an error during the building with Arduino IDE(ESP 2.0.3). After study, this can be fixed with below code in usbhhelp.hpp.

  usb_host_install(&config);
  
    _//Register the client
    usb_host_client_config_t client_config = {
        .is_synchronous = false,
        .max_num_event_msg = 5,
        .async = {
            .client_event_callback = _client_event_callback,
            .callback_arg = Client_Handle,
        }
    };_
    
  err = usb_host_client_register(&client_config, &Client_Handle);
  ESP_LOGI("", "usb_host_client_register: %x", err);

After this, I have tried "descdump", it works well:

8ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
entry 0x403b61d8
[   114][I][usbhhelp.hpp:77] usbh_setup(): [] usb_host_install: 0
[   114][I][usbhhelp.hpp:92] usbh_setup(): [] usb_host_client_register: 0
[   499][I][usbhhelp.hpp:40] _client_event_callback(): [] New device address: 1
[   499][I][usbhhelp.hpp:49] _client_event_callback(): [] speed: 1 dev_addr 1 vMaxPacketSize0 8 bConfigurationValue 1
[   505][I][show_desc.hpp:44] show_dev_desc(): [] bLength: 18
[   511][I][show_desc.hpp:45] show_dev_desc(): [] bDescriptorType(device): 1
[   517][I][show_desc.hpp:46] show_dev_desc(): [] bcdUSB: 0x200
[   523][I][show_desc.hpp:47] show_dev_desc(): [] bDeviceClass: 0x00
[   529][I][show_desc.hpp:48] show_dev_desc(): [] bDeviceSubClass: 0x00
[   535][I][show_desc.hpp:49] show_dev_desc(): [] bDeviceProtocol: 0x00
[   542][I][show_desc.hpp:50] show_dev_desc(): [] bMaxPacketSize0: 8
[   548][I][show_desc.hpp:51] show_dev_desc(): [] idVendor: 0x46d
[   554][I][show_desc.hpp:52] show_dev_desc(): [] idProduct: 0xc52f
[   560][I][show_desc.hpp:53] show_dev_desc(): [] bcdDevice: 0x3000
[   566][I][show_desc.hpp:54] show_dev_desc(): [] iManufacturer: 1
[   571][I][show_desc.hpp:55] show_dev_desc(): [] iProduct: 2
[   577][I][show_desc.hpp:56] show_dev_desc(): [] iSerialNumber: 0
[   583][I][show_desc.hpp:57] show_dev_desc(): [] bNumConfigurations: 1
[   589][I][show_desc.hpp:64] show_config_desc(): [] bLength: 9
[   595][I][show_desc.hpp:65] show_config_desc(): [] bDescriptorType(config): 2
[   602][I][show_desc.hpp:66] show_config_desc(): [] wTotalLength: 59
[   608][I][show_desc.hpp:67] show_config_desc(): [] bNumInterfaces: 2
[   614][I][show_desc.hpp:68] show_config_desc(): [] bConfigurationValue: 1
[   621][I][show_desc.hpp:69] show_config_desc(): [] iConfiguration: 4
[   627][I][show_desc.hpp:74] show_config_desc(): [] bmAttributes(, Remote Wakeup): 0xa0
[   635][I][show_desc.hpp:75] show_config_desc(): [] bMaxPower: 49 = 98 mA
[   642][I][show_desc.hpp:82] show_interface_desc(): [] bLength: 9
[   648][I][show_desc.hpp:83] show_interface_desc(): [] bDescriptorType (interface): 4
[   655][I][show_desc.hpp:84] show_interface_desc(): [] bInterfaceNumber: 0
[   662][I][show_desc.hpp:85] show_interface_desc(): [] bAlternateSetting: 0
[   669][I][show_desc.hpp:86] show_interface_desc(): [] bNumEndpoints: 1
[   675][I][show_desc.hpp:87] show_interface_desc(): [] bInterfaceClass: 0x03
[   682][I][show_desc.hpp:88] show_interface_desc(): [] bInterfaceSubClass: 0x01
[   689][I][show_desc.hpp:89] show_interface_desc(): [] bInterfaceProtocol: 0x02
[   696][I][show_desc.hpp:90] show_interface_desc(): [] iInterface: 0
[   702][I][show_desc.hpp:115] show_hid_desc(): [] bLength: 9
[   708][I][show_desc.hpp:116] show_hid_desc(): [] bDescriptorType (HID): 33
[   715][I][show_desc.hpp:117] show_hid_desc(): [] bcdHID: 0x0111
[   720][I][show_desc.hpp:118] show_hid_desc(): [] bCountryCode: 0
[   726][I][show_desc.hpp:119] show_hid_desc(): [] bNumDescriptor: 1
[... 96 lines cut ...]
[   829][I][show_desc.hpp:86] show_interface_desc(): [] bNumEndpoints: 1
[   835][I][show_desc.hpp:87] show_interface_desc(): [] bInterfaceClass: 0x03
[   842][I][show_desc.hpp:88] show_interface_desc(): [] bInterfaceSubClass: 0x00
[   849][I][show_desc.hpp:89] show_interface_desc(): [] bInterfaceProtocol: 0x00
[   857][I][show_desc.hpp:90] show_interface_desc(): [] iInterface: 0
[   863][I][show_desc.hpp:115] show_hid_desc(): [] bLength: 9
[   868][I][show_desc.hpp:116] show_hid_desc(): [] bDescriptorType (HID): 33
[   875][I][show_desc.hpp:117] show_hid_desc(): [] bcdHID: 0x0111
[   881][I][show_desc.hpp:118] show_hid_desc(): [] bCountryCode: 0
[   887][I][show_desc.hpp:119] show_hid_desc(): [] bNumDescriptor: 1
[   893][I][show_desc.hpp:120] show_hid_desc(): [] bDescriptorType: 34
[   899][I][show_desc.hpp:121] show_hid_desc(): [] wDescriptorLength: 204
[   906][I][show_desc.hpp:100] show_endpoint_desc(): [] bLength: 7
[   911][I][show_desc.hpp:101] show_endpoint_desc(): [] bDescriptorType (endpoint): 5
[   919][I][show_desc.hpp:104] show_endpoint_desc(): [] bEndpointAddress(In): 0x82
[   926][I][show_desc.hpp:107] show_endpoint_desc(): [] bmAttributes(Interrupt): 0x03
[   934][I][show_desc.hpp:108] show_endpoint_desc(): [] wMaxPacketSize: 16
[   940][I][show_desc.hpp:109] show_endpoint_desc(): [] bInterval: 1
[   947][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 1c 15 22 00 00 00
[  1038][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 00 15 22 00 00 00
[  1068][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 00 15 00 00 00 00
[  1078][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 00 00 00 00 00 00
[  1647][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 15 17 1c 00 00 00
[  1768][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 00 00 1c 00 00 00
[  1817][I][usbhhidboot.ino:46] keyboard_transfer_cb(): [] HID report: 00 00 00 00 00 00 00 00

touchgadget, 2022-06-18:
I am glad to hear it is working. I want to remind everyone this code was to help me understand USB and the ESP32 USB host
library https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_host.html. This code does not follow all the ESP32 USB host library recommendations so be cautious. I do not have time to develop this further.

ziv2013, 2022-07-17:
I have used your library to build a USB Keyboard to BLE Keyboard converter. The code is as below. And I'm make a video to show how it works these days. 

`#include <elapsedMillis.h>
#include <BleKeyboard.h>
#include <usb/usb_host.h>
#include "show_desc.hpp"
#include "usbhhelp.hpp"

BleKeyboard bleKeyboard;

bool isKeyboard = false;
bool isKeyboardReady = false;
uint8_t KeyboardInterval;
bool isKeyboardPolling = false;
elapsedMillis KeyboardTimer;

const size_t KEYBOARD_IN_BUFFER_SIZE = 8;
usb_transfer_t *KeyboardIn = NULL;

void keyboard_transfer_cb(usb_transfer_t *transfer)
{
  if (Device_Handle == transfer->device_handle) {
    isKeyboardPolling = false;
    if (transfer->status == 0) {
      if (transfer->actual_num_bytes == 8) {
        uint8_t *const p = transfer->data_buffer;
        ESP_LOGI("", "HID report: %02x %02x %02x %02x %02x %02x %02x %02x",
            p[0], p[1], p[2], p[3], p[4], p[5], p[6], p[7]);
        if(bleKeyboard.isConnected()) {
            bleKeyboard.sendReport((KeyReport*) p);
          }
      }
      else {
        ESP_LOGI("", "Keyboard boot hid transfer too short or long");
      }
    }
    else {
      ESP_LOGI("", "transfer->status %d", transfer->status);
    }
  }
}

void check_interface_desc_boot_keyboard(const void *p)
{
  const usb_intf_desc_t *intf = (const usb_intf_desc_t *)p;

  if ((intf->bInterfaceClass == USB_CLASS_HID) &&
      (intf->bInterfaceSubClass == 1) &&
      (intf->bInterfaceProtocol == 1)) {
    isKeyboard = true;
    ESP_LOGI("", "Claiming a boot keyboard!");
    esp_err_t err = usb_host_interface_claim(Client_Handle, Device_Handle,
        intf->bInterfaceNumber, intf->bAlternateSetting);
    if (err != ESP_OK) ESP_LOGI("", "usb_host_interface_claim failed: %x", err);
  }
}

void prepare_endpoint(const void *p)
{
  const usb_ep_desc_t *endpoint = (const usb_ep_desc_t *)p;
  esp_err_t err;

  // must be interrupt for HID
  if ((endpoint->bmAttributes & USB_BM_ATTRIBUTES_XFERTYPE_MASK) != USB_BM_ATTRIBUTES_XFER_INT) {
    ESP_LOGI("", "Not interrupt endpoint: 0x%02x", endpoint->bmAttributes);
    return;
  }
  if (endpoint->bEndpointAddress & USB_B_ENDPOINT_ADDRESS_EP_DIR_MASK) {
    err = usb_host_transfer_alloc(KEYBOARD_IN_BUFFER_SIZE, 0, &KeyboardIn);
    if (err != ESP_OK) {
      KeyboardIn = NULL;
      ESP_LOGI("", "usb_host_transfer_alloc In fail: %x", err);
      return;
    }
    KeyboardIn->device_handle = Device_Handle;
    KeyboardIn->bEndpointAddress = endpoint->bEndpointAddress;
    KeyboardIn->callback = keyboard_transfer_cb;
    KeyboardIn->context = NULL;
    isKeyboardReady = true;
    KeyboardInterval = endpoint->bInterval;
    ESP_LOGI("", "USB boot keyboard ready");
  }
  else {
    ESP_LOGI("", "Ignoring interrupt Out endpoint");
  }
}

void show_config_desc_full(const usb_config_desc_t *config_desc)
{
  // Full decode of config desc.
  const uint8_t *p = &config_desc->val[0];
  static uint8_t USB_Class = 0;
  uint8_t bLength;
  for (int i = 0; i < config_desc->wTotalLength; i+=bLength, p+=bLength) {
    bLength = *p;
    if ((i + bLength) <= config_desc->wTotalLength) {
      const uint8_t bDescriptorType = *(p + 1);
      switch (bDescriptorType) {
        case USB_B_DESCRIPTOR_TYPE_DEVICE:
          ESP_LOGI("", "USB Device Descriptor should not appear in config");
          break;
        case USB_B_DESCRIPTOR_TYPE_CONFIGURATION:
          show_config_desc(p);
          break;
        case USB_B_DESCRIPTOR_TYPE_STRING:
          ESP_LOGI("", "USB string desc TBD");
          break;
        case USB_B_DESCRIPTOR_TYPE_INTERFACE:
          USB_Class = show_interface_desc(p);
          check_interface_desc_boot_keyboard(p);
          break;
        case USB_B_DESCRIPTOR_TYPE_ENDPOINT:
          show_endpoint_desc(p);
          if (isKeyboard && KeyboardIn == NULL) prepare_endpoint(p);
          break;
        case USB_B_DESCRIPTOR_TYPE_DEVICE_QUALIFIER:
          // Should not be config config?
          ESP_LOGI("", "USB device qual desc TBD");
          break;
        case USB_B_DESCRIPTOR_TYPE_OTHER_SPEED_CONFIGURATION:
          // Should not be config config?
          ESP_LOGI("", "USB Other Speed TBD");
          break;
        case USB_B_DESCRIPTOR_TYPE_INTERFACE_POWER:
          // Should not be config config?
          ESP_LOGI("", "USB Interface Power TBD");
          break;
        case 0x21:
          if (USB_Class == USB_CLASS_HID) {
            show_hid_desc(p);
          }
          break;
        default:
          ESP_LOGI("", "Unknown USB Descriptor Type: 0x%x", bDescriptorType);
          break;
      }
    }
    else {
      ESP_LOGI("", "USB Descriptor invalid");
      return;
    }
  }
}

void setup()
{
  ESP_LOGI("","Starting BLE work!");
  bleKeyboard.begin();
  usbh_setup(show_config_desc_full);
}

void loop()
{
  usbh_task();

  if (isKeyboardReady && !isKeyboardPolling && (KeyboardTimer > KeyboardInterval)) {
    KeyboardIn->num_bytes = 8;
    esp_err_t err = usb_host_transfer_submit(KeyboardIn);
    if (err != ESP_OK) {
      ESP_LOGI("", "usb_host_transfer_submit In fail: %x", err);
    }
    isKeyboardPolling = true;
    KeyboardTimer = 0;
  }

}
`

yanbwang, 2022-11-28:
> Esp32 S3 connected to USB keyboard fail [image]

Could you try another Keyboard and mouse? Don't use the Keyboard which behind the USB HUB.