touchgadget/esp32-usb-host-demos #8: Midi Host issue

LenarOsx, 2022-12-03:
Hey thanks for the code! But i'm getting this error when I try to connect a midi device : 'HUB: Bad transfer status: CHECK_SHORT_DEV_DESC' do you any idea, what could I've done wrong? Thanks.

mdroberts1243, 2023-06-05:
> Hey thanks for the code! But i'm getting this error when I try to connect a midi device : 'HUB: Bad transfer status: CHECK_SHORT_DEV_DESC' do you any idea, what could I've done wrong? Thanks.

I have this problem when trying to connect a Novation Launchpad as well. Wondering if there's been anybody getting past this problem?

mdroberts1243, 2023-06-09:
> I have this problem when trying to connect a Novation Launchpad as well. Wondering if there's been anybody getting past this problem?

My problem was specific to the Novation Launchpad.  It 'resolved' when I plugged an USB Power Meter inline with the Launchpad and that was sufficient for it to enumerate properly.  Still haven't figured out why though.

TrailerParkerBeats, 2024-03-14:
> My problem was specific to the Novation Launchpad. It 'resolved' when I plugged an USB Power Meter inline with the Launchpad and that was sufficient for it to enumerate properly. Still haven't [...]

How was the power meter connected from the usb otg port then to the Launchpad? I am currently trying to connect a Launchpad myself i believed it was some sort of message that is sent between the host and device when connected but still haven’t manage to get it to come out if standby mode when connected to my esp32-s3

ElectroTechnique, 2024-05-09:
I had this working with a Arturia Minilab Mk1. Yes I had a USB power meter connected. Now I get the same message with or without the powermeter connected. The Minilab does send a stream of MIDI messages when connected, but it was handled fine and showed-up in the terminal monitor.