touchgadget/esp32-usb-host-demos #12: ESP32-S3 devkit: no need for the 5V breakout board!

Suxsem, 2023-06-26:
I don't know if this is true also for the ESP32-S**2** devkit, but for the S3 version there is a hidden pad underneath the board the must be shorted out. Maybe it worth to update the readme. Thank you for the great code BTW!

Before:
[image]

After:
[image]

romainreignier, 2023-07-26:
Great! Good to know!
But this seems to be a clone of the DevKit, not the official Espressif [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html), isn't it?

wolsoft, 2023-08-07:
Can i connect flash drive to this usb port to read and write files after shorting this pads?

ludoux, 2023-08-20:
It seems that some 3-rd party dev boards have this feature. 

Accoring to YD-ESP32-S3 dev board's [doc](https://github.com/vcc-gnd/YD-ESP32-S3/blob/main/5-public-YD-ESP32-S3-Hardware%20info/YD-ESP32-S3-SCH-V1.4.pdf), when the USB-ORG is connected, it will let 5V directly pass to VBUS.

[image]

wolsoft, 2023-09-21:
So maybe you have some example code how to write file on usb flash drive connected to this port ?

01GOD, 2024-08-20:
Thanks for explaining what that is for!

marchingband, 2024-08-20:
@01GOD based on your other comments on other repos where you talk about connecting to a DAW, you do not need Host, you need USB Device. USB MIDI Device only requires Arduino MIDI Library, USB examples are in their documentation. If you post a question in Reddit or Arduino forum you would get these answers without spamming the library maintainers.

01GOD, 2024-08-20:
> @01GOD based on your other comments on other repos where you talk about connecting to a DAW, you do not need Host, you need USB Device. USB MIDI Device only requires Arduino MIDI Library, USB [...]

I am not a redditor. I had tried the Arduino MIDI Library, but the examples didn't function as expected. I would like to add USB MIDI in from DAW to a capacitive touch synthesizer project, and also to an S3 PCB that I designed and assembled. Surprised didn't tell Me to buy a wvr board, as were telling others to when they asked questions.

marchingband, 2024-08-20:
Then you need USB Device, not USB Host.
https://support.minidsp.com/support/solutions/articles/47001243548-understanding-the-difference-between-usb-host-and-usb-device-mode
When you do use the S3 usb port for a function like this, you will no longer be able to use the USB port for logging, it can only do 1 thing at a time, so you will have to research and learn how to get logs another way.
If you have further questions I would encourage you to use the Arduino Forum if you don't like reddit, your question has nothing to do with this library.

01GOD, 2024-08-21:
> @01GOD Then you need USB Device, not USB Host. https://support.minidsp.com/support/solutions/articles/47001243548-understanding-the-difference-between-usb-host-and-usb-device-mode When you do use the S3 usb port for a function like this, you will no longer be able to use the USB port for logging, it can only do 1 thing at a time, so you will have to research and learn how to get logs another way. If you have further questions I would encourage you to use the Arduino Forum if you don't like reddit, your question has nothing to do with this library.

The board I have has 2 USB ports. 

This is an opportunity to do a brocode flex if have 1337 skills, by posting an example that DEFINITELY FUNCTIONS. Consider that a marketing opportunity to post a small repo with a fully functional example of laptop DAW to ESP32-S3 (without any middleman) so that popular ESP32 synth community repos (such as the Marcel License synths) don't need to use a separate IC/board for USB MIDI. 

Are 1337 enough? Are ya?