touchgadget/esp32-usb-host-demos #16: Don't work when MIDI Device re-plug

oujianshen, 2024-04-13:
### This is a great project, I can run it and all the functions are normal
When the MIDI Device is reinserted, it cannot be automatically recognized and needs to be restarted to function properly again
I see that the new device is defined in the function in usbhelp.hpp as void _cliend_event_callback (const usbhost_cliend_event_msg_t * event_msg, void * arg)
I guess this function was not called during the second insertion But my ability is limited and I can't change this code well Who can help fix this issue of reinsertion and rediscovery?
Thank you very much

fastfourier666, 2024-04-20:
Looks like there needs to be some extra work done after the device is unplugged. According to [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_host.html) in the Clients & Class driver -> lifecycle section. 

I managed to get the enumeration to happen on replugging, but in the case of the MIDI example you have to re-establish the endpoints as well.

fastfourier666, 2024-04-20:
I made a hackjob fix [here](https://github.com/fastfourier666/esp32_usb_midi_host_fixed)

The original does not do any cleanup when the device is unplugged - either in the sketch or the ESP USB Host library. I added a callback so usbhelp.hpp can call a function in the main sketch when the device is "gone".

I'm sure it is pretty janky, would love to hear some improvements. I know next to nothing about USB. Cheers.

qa2080639, 2024-07-25:
>  ~~ I spent a whole day looking for this problem and finally found the reason. 

use `usb_host_interface_release`,`usb_host_device_close` can fix it
example code `usbhhelp.hpp`
```
case USB_HOST_CLIENT_EVENT_DEV_GONE:
    ESP_LOGI("", "Device Gone handle: %x", event_msg->dev_gone.dev_hdl);
    err = usb_host_interface_release(Client_Handle, event_msg->dev_gone.dev_hdl, 0);
    if (err != ESP_OK)
      ESP_LOGI("", "usb_host_interface_release: %x", err);
    err = usb_host_device_close(Client_Handle, event_msg->dev_gone.dev_hdl);
    if (err != ESP_OK)
      ESP_LOGI("", "usb_host_device_close: %x", err);
    break;
```

01GOD, 2024-08-20:
> case USB_HOST_CLIENT_EVENT_DEV_GONE:
>     ESP_LOGI("", "Device Gone handle: %x", event_msg->dev_gone.dev_hdl);
>     err = usb_host_interface_release(Client_Handle, event_msg->dev_gone.dev_hdl, 0); [...]

Can explain how? I tried it and I know of a skilled embedded engineer that also tried it. 

When I flashed an ESP32-S3 dev kit clone with that in verbose mode, it simply flashes it and the transmit LED turns on, but no serial monitor messages sadly. 

Recommended solutions? Thanks in advance!

EParisot, 2024-09-13:
@fastfourier666 You did a great job thank you for sharing !