touchgadget/esp32-usb-host-demos #17: Configuration descriptor larger than control transfer max length

hdsjulian, 2024-11-18:
With one MIDI Keybaord everything works. With the other i get: 
E (954) HUB: Configuration descriptor larger than control transfer max length
E (954) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
[  1292][I][usbhhelp.hpp:108] usbh_task(): [] No more devices

The Keyboard that works is an LPK25, the other registers as NC2-CONTROLLER. 
I somehow first assumed it might have to do with USB_HID_DESC_SIZE in show_desc.hpp, but that is not the case. Changing the value to 24 instead of 9 didn't do the trick. 
My knowledge of USB and everything is not enough. Any ideas what could help me? Is there a good way of debugging it?

serifpersia, 2024-12-10:
> With one MIDI Keybaord everything works. With the other i get: E (954) HUB: Configuration descriptor larger than control transfer max length E (954) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC [ [...]

Some MIDI devices have big short descriptor(default is 256) and for some midi devices this is not big enough. The fix is to custom compile esp32 arduino core sdk with new value for this parameter(I compiled mine with max 4096 bytes). You can find gdrive link in my esp32 [project](https://github.com/serifpersia/pianolux-esp32/blob/main/ESP32_PianoLux/ESP32_PianoLux.ino#L21) if you want to quickly fix this issue(s2/s3 sdks)

civet, 2025-10-30:
@serifpersia Could you tell me how to recompile the SDK? I have tried many times but failed. Whether in a Linux environment or Docker, errors always occur. Which version is the usable one?

I am using PlatformIO, and it seems that the SDK is not the latest, so I am not sure if replacing the files will work.